ros-drivers/urg_node #18: ARM vs X86

synologic, 2016-01-30:
Running ROS Indigo on ARM (Odroid XU4) i cannot get any data from the scanner:

[ INFO] [1454179348.727796454]: Connected to network device with intensity and ID: H1513161
[ INFO] [1454179348.801499047]: Streaming data.
[ WARN] [1454179350.111422222]: Could not grab single echo scan.
[ERROR] [1454179350.893385932]: Error count exceeded limit, reconnecting.
[ERROR] [1454179353.900147769]: Unknown error connecting to Hokuyo

While, the same scanner, same network, same settings, works fine on the X86.

Any insight on this ?

marcinbogdanski, 2016-03-27:
Hi

I'm not sure if this will be helpful in any way, I'm have a problem with urg_node as well. My setup is RPi2, Ubuntu 14.04 with UST-20LX scanner. It works out of the box with urg_node installed from ubuntu repo, but if I pull Ethernet cable out and back in it will not reconnect.

I think urg_node error actually originates in URGCWrapper constructor [here](https://github.com/ros-drivers/urg_node/blob/indigo-devel/src/urg_c_wrapper.cpp). Which in turns comes from underlying urg_c package funtion urg_open [here](https://github.com/ros-drivers/urg_c/blob/master/current/src/urg_sensor.c). 

Hmm....

mikeferguson, 2025-05-22:
Relates to EOL ROS/Ubuntu