ros-drivers/urg_node #21: Request to release for ROS Kinetic

130s, 2016-09-08:
Thanks!

Btw I see it's released for Jade so #11 should be closed.

doisyg, 2016-09-09:
+1 
a release for kinetic would be really nice

130s, 2016-09-23:
@tfoote could you take care of this? It's been 2weeks+ and no response is heard.

tfoote, 2016-09-23:
I tried blooming it but this depends on laser_proc which is not yet released.

kmhallen, 2016-09-26:
We might want to merge #13 first.

schvarcz, 2017-01-23:
Just a input for this issue.

I am using the pkg available at ubuntu repository for kinetic (sudo apt-get install ros-kinetic-urg-node) for get data from my Hokuyo URG 04LX and it is workinng as a charm.

So maybe they do not put the code for kinetic in this repository, maybe they are using the indigo branch or maybe my computer is a time travel machine that is able to download code not yet available!!!! 

I just know it is working great with ROS Kinetic. So, as they took time to close the same issue for ROS Jade, I strongly think that they solve the issue here, but did not close it yet....

For the next souls looking for this pkg to ROS, it is already available.

kmhallen, 2017-01-23:
Yes, there is a release for Kinetic.   
https://discourse.ros.org/t/new-packages-for-kinetic-2016-10-04/544

However, Kinetic users still do not have access to the ```getID``` program. PR #13 has not been merged, and the [hokuyo_node](http://wiki.ros.org/hokuyo_node) package that also has getID is EOL.

tonybaltovski, 2017-03-21:
Released!

closed 2017-03-21 by tonybaltovski