ros-drivers/urg_node #23: Problems with Hokuyo 04LX

boris-il-forte, 2016-12-12:
Hi,
I'm trying to use the URG-04LX sensor, with the new urg_node. When I try to connect I get:
"Unknown error connecting to Hokuyo."

While, if I'm using the old hokuyo_node, the sensor just work correctly.
Also, if I try to connect using the urg_node after killing the hokuyo node, the new node works.

 I've tried with different parameters and disabling/enabling calibration

Also I've updated the sensor to the latest version (Ver3.4.03) without success.

Restarting the sensor doesn't work, as it enters in a loop of error states.

boris-il-forte, 2016-12-21:
Update: using the verison in the pull request #22 i get a more informative error:

$ rosrun urg_node urg_node 
[ERROR] [1482329968.746021512]: Error connecting to Hokuyo: Could not initialize Hokuyo:
not connected.

$ rosrun urg_node getID /dev/ttyACM0
getID failed: Could not initialize Hokuyo:
not connected.

However the hokuyo is connected and in fact works with the old hokuyo_node:
$ rosrun hokuyo_node hokuyo_node 
[ INFO] [1482330147.593798394]: Connected to device with ID: H0707625
[ INFO] [1482330147.692804670]: Starting calibration. This will take up a few seconds.
[ INFO] [1482330150.783847053]: Calibration finished. Latency is: -0.0810
[ INFO] [1482330151.093119302]: Streaming data.

the hokuyo is on the default serial over usb port:
$ ls /dev/ttyACM*
/dev/ttyACM0

bygreencn, 2017-03-29:
please check pull-request of https://github.com/ros-drivers/urg_c/pull/4

2scholz, 2017-09-22:
I ran into the same problem and found a solution. 
There are two different communication protocols the URG-04LX can use: SCIP1.1 and SCIP2.0. 
The standard default mode for the URG-04LX is SCIP1.1, but the urg_node does not support this mode. 

The solution is to switch to SCIP2.0. 
This can be done by sending a command to the sensor, for example using picocom: 
`sudo picocom /dev/ttyACM0`
Then type `SCIP2.0` and press enter. 

This has to be done at every startup though. 

A more convenient method I haven't tested is to set the default mode to SCIP2.0. 
Details are in this [datasheet](https://en.manu-systems.com/URG_SCIP20.pdf) on page 3.

closed 2017-10-17 by tonybaltovski

JanezCim, 2020-04-03:
I had a similar problem when I was trying it with urg_node installed from binaries. I solved it simply by removing the binaries, and compiling the latest ROS package from urg_node github page https://github.com/ros-drivers/urg_node

PR #35 (2017-10-17): Add backwards-compatible support for the Hokuyo URG-04LX
`urg_node` does not support SCIP 1.1, only SCIP 2.0.
However, the Hokuyo URG-04LX supports both SCIP 1.1 and SCIP 2.0 and usually defaults to SCIP 1.1.
It needs to be switched to SCIP 2.0 at every startup.

For this purpose the function `URGCWrapper::setToSCIP2` was added.
A Hokuyo URG-04LX in SCIP 1.1 mode lead to an exception in `URGCwrapper::initialize`. 
Now, before throwing the exception an attempt to switch the sensor to SCIP 2.0 mode is made using `URGCWrapper::setToSCIP2`. 

The `hokuyo_node` provides support for the URG-04LX in  [... cut]