ros-drivers/urg_node #30: The new AR00 function breaks urg_node

tipara, 2017-05-18:
Hello,

It seems that the recent commit "Add getAR00 status command" breaks urg_node, at least using my version of the UST-20LX over ethernet. I suspect #29 (maybe) and [http://answers.ros.org/question/261367/problem-with-running-urg_node-with-ust-20lx/](url) (almost surely) are issues related to this problem.

How to reproduce (tried on 3 different computers, with the same Hokuyo unit): on a fresh install of Kinetic on 16.04, clone urg_c, urg_node and laser_proc, then catkin_make. (I got the same problem with the apt install.)
The command:
`rosrun urg_node urg_node _ip_address:="192.168.0.10"`
fails with the following message:
`Connected to network device with intensity and ID: H1528xxx`
`  terminate called after throwing an instance of 'std::length_error'`
`  what(): basic_string::_M_create`
`Aborted`

On Ubuntu 14, I get the same message with basic_string::_S_create instead of M.

Temporary f0ix: in file   urg_node_driver.cpp, comment the line (number 466):
`    // Before starting, update the status`
`    updateStatus();`

My comments:
Using Wireshark, it seems the sensor does not reply to the AR00 command, which causes the call in line 555 of urg_c_wrapper.cpp
`    read_len = read(sock, recvb + total_read_len, expected_read - total_read_len);` 
in URGCWrapper::sendCommand to return -1. However, since read_len is unsigned, it loops back to MAX_INT and so the verification:
`    if (read_len <= 0)`
is never triggered, causing a crash on the line:
`  std::string recv_header(recvb, read_len);`

Looking at the Hokuyo documentation, I see no reference to an "AR" command, which could explain the lack of response from the sensor.

Hope this will help

1chick, 2017-06-06:
also comment the function of sending AR00 command , I m do that copy this problem.

tonybaltovski, 2017-06-06:
@tipara @1chick could you verify that b3f98217afebfeae232a67b429426e446f797d0b solves your problem?

tonybaltovski, 2018-04-14:
Closing due to inactivity.  Please feel free to reopen if the issue persists.

closed 2018-04-14 by tonybaltovski

amaldo, 2019-10-11:
I would like to comment that I have the same issue on a UST-10LX laser with the most recent firmware (2.23). Using https://github.com/ros-drivers/urg_node/commit/b3f98217afebfeae232a67b429426e446f797d0b fixes the problem for me.