ros-drivers/urg_node #32: can't change the hokuyo utm-30lx-ew ip addess using set_urg_ip.py

le-duckietown, 2017-06-19:
Hi,

I tried to change the IP address for my hokuyo utm-30lx-ew using the set_urg_ip.py provided in the package but it didn't work, the current IP still stayed the same. Do you know what problem is this?

tonybaltovski, 2017-06-20:
Was there an error when you ran the command?  Which version of urg_node are you using?

le-duckietown, 2017-06-20:
There was no error when I ran the command. When I requested to change to the new IP, the terminal said "Connecting to 192.168.0.10", which is my default, old IP, then "Updating settings. Done updating, cycle power on laser". However, nothing happened, I can still use only the default IP and not the new one

My urg_node version is 0.1.10.