ros-drivers/urg_node #34: CMake Error

Yasermn, 2017-09-18:
When I copied urg_node-groovy-devel package into my catkin_ws/src folder and tried to compile it, I got the following errors:

CMake Error at /opt/ros/groovy/share/catkin/cmake/catkinConfig.cmake:75 (find_package): Could not find a configuration file for package urg_c.
Set urg_c_DIR to the directory containing a CMake configuration file for urg_c. The file will have one of the following names:
urg_cConfig.cmake
urg_c-config.cmake
Call Stack (most recent call first): urg_node-groovy-devel/CMakeLists.txt:7 (find_package)
CMake Error at /opt/ros/groovy/share/catkin/cmake/catkinConfig.cmake:75 (find_package): Could not find a configuration file for package laser_proc.
Set laser_proc_DIR to the directory containing a CMake configuration file for laser_proc. The file will have one of the following names:
laser_procConfig.cmake
laser_proc-config.cmake
Call Stack (most recent call first): urg_node-groovy-devel/CMakeLists.txt:7 (find_package)
-- Configuring incomplete, errors occurred! Invoking "cmake" failed

Could you please tell me what is going wrong? And How can I solve this?

tonybaltovski, 2018-04-14:
You need to have `urg_c` and `laser_proc` installed since they dependencies of this package.  Use `rosdep` to install them.

mikeferguson, 2025-05-22:
Groovy is EOL

closed 2025-05-22 by mikeferguson