ros-drivers/urg_node #37: urg_node failed on Intel NUC / swapping issue

FaboNo, 2017-11-20:
I am facing a problem running urg_node with an Hokuyo 10LX on an embedded computer.

Here is the environment, I am running ROS kinetics on Ubuntu 16.06LTS on an INTEL NUC 5ITRYH with 8GO DDR4 and SSD 200Go.

When I start urg_node, it works perfectly but as soon as I open a terminal for instance, I got the following warning:

process[urg_node-2]: started with pid [5099] [ INFO] [1511171038.966908047]: Connected to network device with ID: H1642740 [ INFO] [1511171038.979387775]: Starting calibration. This will take a few seconds. [ WARN] [1511171038.979419672]: Time calibration is still experimental. [ INFO] [1511171039.982890964]: Calibration finished. Latency is: -0.0208. [ INFO] [1511171039.998174108]: Streaming data. [ WARN] [1511171050.027225044]: Could not grab single echo scan. [ERROR] [1511171051.692534732]: Error count exceeded limit, reconnecting. [ INFO] [1511171053.704590758]: Connected to network device with ID: H1642740 [ INFO] [1511171053.716937334]: Starting calibration. This will take a few seconds. [ WARN] [1511171053.717063649]: Time calibration is still experimental. [ INFO] [1511171054.718481924]: Calibration finished. Latency is: -0.0212. [ INFO] [1511171054.734664754]: Streaming data.

And it if I start a slam with RVIZ, i finally ended up with:

[ WARN] [1511170394.929138196]: Time calibration is still experimental. [FATAL] [1511170395.851344341]: Could not calibrate time offset: Cannot get scan to measure time stamp offset. [urg_node-2] process has died [pid 4389, exit code -5, cmd /home/fabrice/catkin_demo_lidar_ws/devel/lib/urg_node/urg_node __name:=urg_node __log:=/home/fabrice/.ros/log/c61d770e-cdd5-11e7-b1f1-94c691132292/urg_node-2.log]. log file: /home/fabrice/.ros/log/c61d770e-cdd5-11e7-b1f1-94c691132292/urg_node-2*.log

It looks like, at first glance, that the NUC takes too much time to swap between tasks but I am I have to admit that I am surprise.

I tried to change the priority between tasks with renice, but without success.

The same programs (urg_node, hector slam...) work perfectly on my laptop with ROS kinetics/Ubuntu 16.04LTS

Does anybody had the same experience and can share how to handle it?

Thank you very much

Fabrice

tonybaltovski, 2018-01-05:
Are you using the driver from debs or source?

FaboNo, 2018-01-06:
@tonybaltovski actually looks like it is an issue with the NUC because we are using a USB2Ethernet and it works fine. Thank you for your message

closed 2018-01-06 by tonybaltovski