ros-drivers/urg_node #38: connecting multiple hokuyo UTM-30LX

jangtu052, 2017-11-23:
hey guys, i am having trouble connecting multiple hokuyo sensors.
I tried various type of launch files. 
down below isthe slightly modified default launch file

`<launch>

<group ns="left">
  <node name="urg_node" pkg="urg_node" type="urg_node" output="screen">
    <param name="ip_address" value=""/>
    <param name="serial_port" value="/dev/ttyACM0"/>
    <param name="serial_baud" value="115200"/>
    <param name="frame_id" value="laser"/>
    <param name="calibrate_time" value="false"/>
    <param name="publish_intensity" value="true"/>
    <param name="publish_multiecho" value="false"/>
    <param name="angle_min" value="-1.5707963"/>
    <param name="angle_max" value="1.5707963"/>
    <remap from="scan" to="left_scan"/>	
  </node>
</group>

<group ns="right">
  <node name="urg_node" pkg="urg_node" type="urg_node" output="screen">
    <param name="ip_address" value=""/>
    <param name="serial_port" value="/dev/ttyACM1"/>
    <param name="serial_baud" value="115200"/>
    <param name="frame_id" value="laser"/>
    <param name="calibrate_time" value="false"/>
    <param name="publish_intensity" value="true"/>
    <param name="publish_multiecho" value="false"/>
    <param name="angle_min" value="-1.5707963"/>
    <param name="angle_max" value="1.5707963"/>
    <remap from="scan" to="right_scan"/>
  </node>
</group>

</launch>`

then second one runs for a instance and makes an error. like down below

[ INFO] [1511419973.276698778]: Streaming data.
[ WARN] [1511419974.989649912]: Could not grab single echo scan.
[ERROR] [1511419978.506572805]: Error count exceeded limit, reconnecting.
[ERROR] [1511419984.335448520]: Error connecting to Hokuyo: Could not initialize Hokuyo:
not connected.
[ERROR] [1511419996.835947865]: Error connecting to Hokuyo: Could not initialize Hokuyo:
not connected.

Any ideas??

tonybaltovski, 2018-01-05:
Are you able to connect to each unit individually at `/dev/ttyACM0` and `/dev/ttyACM1`?

EchChou, 2019-10-10:
hi everyone
I have the same problem then "jangtu052". I can't connect to multiple hokuyo.
I'm using the next launche file:
<launch>
    <node ns="hokuyo0" pkg="urg_node" type="urg_node" name="hokuyo0">
        <param name="port" type="string" value="/dev/ttyACM0" />
        <remap from="scan" to="scan0" />
    </node>

    <node ns="hokuyo1" pkg="urg_node" type="urg_node" name="hokuyo1" >
        <param name="port" type="string" value="/dev/ttyACM1" />
        <remap from="scan" to="scan1" />
    </node>
</launch>

I get the next error : 

started roslaunch server http://localhost:44229/

SUMMARY
========

PARAMETERS
 * /hokuyo0/hokuyo0/port: /dev/ttyACM0
 * /hokuyo1/hokuyo1/port: /dev/ttyACM1
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /hokuyo1/
    hokuyo1 (urg_node/urg_node)
  /hokuyo0/
    hokuyo0 (urg_node/urg_node)

ROS_MASTER_URI=http://localhost:11311

process[hokuyo0/hokuyo0-1]: started with pid [10037]
process[hokuyo1/hokuyo1-2]: started with pid [10038]
[ERROR] [1570697719.232256864]: Error connecting to Hokuyo: Could not open serial Hokuyo:
/dev/ttyACM0 @ 115200
invalid response.
[ERROR] [1570697719.531241951]: Error connecting to Hokuyo: Could not open serial Hokuyo:
/dev/ttyACM0 @ 115200
invalid response.
[ERROR] [1570697725.406083508]: Error count exceeded limit, reconnecting.
[ERROR] [1570697729.688121747]: Error connecting to Hokuyo: Could not open serial Hokuyo:
/dev/ttyACM0 @ 115200
invalid response.
[ERROR] [1570697733.871094387]: Error count exceeded limit, reconnecting.
[ERROR] [1570697736.260121162]: Error connecting to Hokuyo: Could not open serial Hokuyo:
/dev/ttyACM0 @ 115200
invalid response.

Someone can help please.
Thank you.

mikeferguson, 2025-05-23:
No response from OP

closed 2025-05-23 by mikeferguson