ros-drivers/urg_node #40: ROS 2 support

bponsler, 2017-12-22:
There is a fork of urg_node (https://github.com/bponsler/urg_node) that ports urg_node to ROS 2. @Karsten1987 had a great suggestion of trying to push the ros2 changes upstream so everything is located here.

What are your thoughts on doing this and how best to go about doing it? There are several repos that had to be ported and the messages  were split it to be a separate package.

tonybaltovski, 2018-01-05:
Thanks!  I am actually not sure. I see that is there is ros2 org but there is not ros2-drivers.  Maybe @tfoote could provide some insight?

tfoote, 2018-01-05:
In general we are working with the goal that ROS1 and ROS2 codebases will be closely alligned and not require forking all development and that ROS2 versions can be maintained on existing repositories for most packages that aren't part of the core that's being switched out between ros1 and ros2. I'd suggest that the maintainers here create an `ardent-devel` branch that the ROS2 changes can be PRd against from @bponsler's fork.

tonybaltovski, 2018-01-05:
Thanks Tully, I made the `ardent-devel` branch.

aarushsesto, 2018-05-10:
Hi, I wanted to check for the status of this ros2 wrapper, is it functional yet or has it been tested with the hokuyo lasers? 
If it's still missing something I would certainly like to help, as I need to use it.

bponsler, 2018-05-12:
@aarushsesto: I haven’t had time to work on this in many months, but I had tested this branch with both a hokuyo emulator and actual hardware. I believe other people were successfully using it as well but haven’t heard any news in a while.

Any help you want to provide would be greatly appreciated. Likely it is out of date with recent changes in ROS2.

aarushsesto, 2018-05-13:
So I tried to build the package in ROS2, at first it was throwing an error that the build type does not exist after which I added the `build_type` as `ament_cmake` under the `export` tag in the package manifest file. But Now it's throwing this error:

```
Traceback (most recent call last):
  File "/home/batman/ros2_ws/build/urg_node/catkin_generated/generate_cached_setup.py", line 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python
  "/home/batman/ros2_ws/build/urg_node/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:186 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/batman/ros2_ws/build/urg_node/CMakeFiles/CMakeOutput.log".
See also "/home/batman/ros2_ws/build/urg_node/CMakeFiles/CMakeError.log".
Makefile:1852: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

<== Command '. /home/batman/ros2_ws/build/urg_node/cmake__build.sh && /usr/bin/make cmake_check_build_system' failed in '/home/batman/ros2_ws/build/urg_node' with exit code '2'
<== Command '. /home/batman/ros2_ws/build/urg_node/cmake__build.sh && /usr/bin/make cmake_check_build_system' failed in '/home/batman/ros2_ws/build/urg_node' with exit code '2'

```
I have only sourced `locale_setup.*` files from my ros2 workspace, hence I cannot understand how is the Cmake error is referring to my kinetic distro(which is unsourced)
Also I'm not able to understand the error completely, it seems that the path to catkin environment is missing?
@bponsler can you help me understand the error?

bponsler, 2018-05-13:
It seems like it is still referencing catkin somewhere. Based on the references to catkin and the fact that you had to add the export for ament_build I think you’re not using the correct branch.

The branch that has been updated to work for ROS2 is: https://github.com/bponsler/urg_node/tree/ros2-devel

Is this the fork and branch you are using?

aarushsesto, 2018-05-15:
Thanks for pointing that out, I was actually trying to use the ardent-devel branch from here : [https://github.com/ros-drivers/urg_node/tree/ardent-devel](https://github.com/ros-drivers/urg_node/tree/ardent-devel)
After getting the branch from your repo, I tried compiling but I had a lot of missing dependencies, most of which you have as your repos (thanks!), but I could not find how to install `urg_c` dependency. I tried getting the zip file from source forge, installing it with make install and setting the `CMAKE_PREFIX_PATH` to install source but it doesn't work, do you know how I can fix this? This is the error :
```

CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findurg_c.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "urg_c", but
  CMake did not find one.

  Could not find a package configuration file provided by "urg_c" with any of
  the following names:

    urg_cConfig.cmake
    urg_c-config.cmake

  Add the installation prefix of "urg_c" to CMAKE_PREFIX_PATH or set
  "urg_c_DIR" to a directory containing one of the above files.  If "urg_c"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/batman/ros2_ws/build/urg_node/CMakeFiles/CMakeOutput.log".

```

Also do you know if I can use rosdep here to install some of the dependencies, according to the discussion [here](https://discourse.ros.org/t/downloading-dependencies/1489), I understand I can use rosdep with ros2 as well.

bponsler, 2018-05-16:
You should be able to use urg_c from here: https://github.com/bponsler/urg_c/tree/ros2-devel (be sure to use the ros2_devel branch). Again this is quite old but was working on ros2 at some point.

At the time I was working on this last year I documented all the ros2 packages I was using here: https://github.com/bponsler/ros2-support/blob/master/README.md

So when missing a dependency try the package from this repo - most should have the ros2_devel branch.

As for rosdep I don’t know. At the time I wasn’t using it but it may be possible - you’ll have to ask someone else.

Let me know if you run into additional problems. Also - you should create PRs for whatever changes were necessary to get this working again!

aarushsesto, 2018-05-16:
Thanks for all the help!
Up untill now I haven't made any changes, but if you don't mind I would certainly like to make a PR for a more detailed readme, so it would be easier to use this package in future.

I was able to compile it but now I'm having problem running, I tried to use `ros2 run urg_node urg_node` which gave this output : 

`
/home/batman/ros2_ws/install/lib/urg_node/urg_node: error while loading shared libraries: libliburg_c.so: cannot open shared object file: No such file or directory`

I also tried using the launch command `launch install/share/urg_node/launch/urg_node_launch.py ` which gave this output(without any --args parameter) :

```
Traceback (most recent call last):
  File "/home/batman/ros2_ws/install/bin/launch", line 9, in <module>
    load_entry_point('launch', 'console_scripts', 'launch')()
  File "/home/batman/ros2_ws/build/launch/launch/main.py", line 50, in main
    load_launch_file(launch_file, launch_descriptor, args.args)
  File "/home/batman/ros2_ws/build/launch/launch/loader.py", line 21, in load_launch_file
    launch_module.launch(launch_descriptor, args)
  File "install/share/urg_node/launch/urg_node_launch.py", line 22, in launch
    if name in argv:
TypeError: argument of type 'NoneType' is not iterable

``` 
But when I did `launch install/share/urg_node/launch/urg_node_launch.py --args serial-port` : 
```
(0) pid 2247: ['/home/batman/ros2_ws/install/lib/urg_node/urg_node'] (stderr > stdout, all > console)
[0] /home/batman/ros2_ws/install/lib/urg_node/urg_node: error while loading shared libraries: libliburg_c.so: cannot open shared object file: No such file or directory
(0) rc 127
() tear down
``` 
Since `--args` is for optional arguments, it should not throw exception if missing, I think this can be avoided by just having a check in the python launch file, I'll make a PR for that.

Meanwhile can you tell me what I am doing wrong, or how did you run it before?

bponsler, 2018-05-16:
It looks like you’re getting close.

I think the issue with —args is likely that I never updated the urg_node launch script after this PR was merged: https://github.com/ros2/launch/pull/57

You’re right a check should take care of it. Also, I’d be happy to merge any PRs you create on my various forks and I’m sure many others will appreciate your work.

I was running it from the launch file at the time but that was a long time ago so any number of things could be different and cause issues.

For some reason it’s not able to find the liburg_c library...perhaps it’s not being installed into the correct directory? I think your best path forward is to confirm that both urg_c and urg_node CMakeLists.txt and package.xml files are up to date with typical ros2 use. There may be something I was doing that has changed convention? 

Also, I’d check and confirm that the urg_c library is being created during your build and where it’s getting put. It may help you track it down to understand if it’s being put in a weird spot compared to other packages.

Also - @Karsten1987 may be able to help answer how he was running it since he’s the one who graciously updated urg_node for ament support.

aarushsesto, 2018-05-21:
@bponsler I think I have it in the running state now, you mentioned that you have also tested this with a hokuyo emulator, I couldn't find any such thing online, can you tell me where can I find it, since I also plan to test this driver without the actual hardware first.

bponsler, 2018-05-22:
@aarushsesto Awesome - nice job. 

Unfortunately, the emulator was something developed for a previous employer that is not open source and I don’t have access to anymore. If you end up developing your own it would be greatly appreciated if you release it for others to use!

MarcTestier, 2018-10-16:
Hi,
I'm porting the package to ROS 2 Bouncy based on @bponsler version, adding some of the features that were present in ROS 1 like the reconfiguration of the parameters.

You can find the port [here](https://github.com/hopetechnik/urg_node/tree/ros2-devel).

You can find a status of the feature that I identified, ported and tested in the ROS 1 package [here](https://docs.google.com/spreadsheets/d/1_0kXQhPuvQx3-LFibtIpJFCw35aMaCnop9416IAVZ4U/edit?usp=sharing).

Basically, the main subjects of the port (tell me if I missed something) are : 
- the core to publish the laser data,
- the reconfiguration of the parameters,
- the dependencies related features.

I could only test with a UST-20LX, which is an Ethernet model without multiecho and intensity features and so far : 
- the core features seem okay
- I added the reconfiguration of the parameters by using parameters event but I don't know if that's really what's supposed to replace the ROS 1 `dynamic_reconfiguration` pkg. For now I don't check the given parameter values to enforce correct values (like a limit on the angle, error_limit, etc) , but I'm going to add this using `register_param_change_callback` (see an example [here](https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/parameters/even_parameters_node.cpp)).
I also didn't add every parameters as reconfigurable as I'm still not 100% sure which one reconfigurable in ROS 1.
- I'm trying to move the dependencies from the ports made by @bponsler like the `ros2_time`, `diagnostics`, `ros2_console` and `laser_proc` to the "official" port.

I already moved the `urg_node` package from `ros2_time` to the official implementation of time, clock, etc but the package is still needed for the `diagnostics` pkg.
The `diagnostics` and `ros2_console` port are being discussed by other people respectively [here](https://discourse.ros.org/t/diagnostic-aggregator-and-diagnostic-updater-porting-to-ros2/6382) and [here](https://discourse.ros.org/t/diagnostic-aggregator-and-diagnostic-updater-porting-to-ros2/6382), so I guess I'm going to wait for it to be port and then use them.
Haven't checked how is the port of the `laser_proc` pkg or if there is "official" work on it.

You can have more information in the file shared above, if you have any input or comment, please feel free to share.
Also, should I do a pull request of this so we can have the package in this repo ?

SteveMacenski, 2018-10-16:
Opening a PR to have a ros2 branch would be really helpful for maintainability and visibility reasons

MarcTestier, 2018-10-17:
I'm not used to PR, should we ask @tonybaltovski to create a ros2-devel or should I PR against a pre-existing branch like ardent ?

tonybaltovski, 2018-10-17:
I'm for a PR against a ros2-devel.  Does it work with ardent or only bouncy?

MarcTestier, 2018-10-18:
I have a bouncy version that I made a bit after the release but there is no parameter reconfiguration, it was just a simple port of the ardent version.
The version I have now is made using the master sources so kind of half bouncy, half crystal ? I'm not 100% sure whether or not I can use it on the bouncy debian pkg.

tonybaltovski, 2018-10-23:
Thanks for checking.  I made a Bouncy branch.

MarcTestier, 2018-10-24:
I did a PR : https://github.com/ros-drivers/urg_node/pull/46

Karsten1987, 2019-06-13:
Can we generally have two ROS2 related branches? One for `dashing-devel` and one for `ros2-devel`? I've opened a similar issue here: https://github.com/ros-drivers/urg_c/issues/10

We have a dashing port for the `urg_node` here and would like to have it merged into this upstream repo: https://github.com/bosch-robotics-na/urg_node/tree/fix_ros2

tonybaltovski, 2019-06-13:
I added the branches.  Thanks for the ROS2 port! If @MarcTestier could transfer the repo it would be great!

MarcTestier, 2019-06-14:
I don't mind transferring the repo, which repo do you want me to transfer to ?

Karsten1987, 2019-06-16:
@MarcTestier it would be great if you could transfer your `urg_node_msgs` package to this org unit.

MarcTestier, 2019-06-17:
Hi, I can't transfer the ownership on this org unit as I "don’t have the permission to create repositories on ros-drivers". @Karsten1987 may I transfer it to you instead ?

Karsten1987, 2019-06-17:
I am okay with hosting the repo for a bit, but I believe that's just one more indirection. Ultimately, all urg related packages should land inside the ros-drivers org. 
@tonybaltovski can you create an empty `urg_node_msgs` package and we can open PRs against that repo?

MarcTestier, 2019-06-24:
I agree, no point just moving the repo unless we move it to someone that can create a repo on ros-drivers. I can either transfer it to @tonybaltovski or just wait to open a PR ?

tonybaltovski, 2019-06-24:
@MarcTestier you should just be able to transfer to to ros-drivers.

MarcTestier, 2019-06-25:
@tonybaltovski If I'm not mistaking, we are talking about the "**Transfer ownership**" function right ? If so, as stated above, I get the message `You don’t have the permission to create repositories on ros-drivers`.

Karsten1987, 2019-06-25:
@tonybaltovski I believe the easiest way is when you create a new repo `urg_node_msgs` and initialize it with some `readme.md`. That would allow @MarcTestier to just open a PR against it.

tonybaltovski, 2019-06-27:
Created: https://github.com/ros-drivers/urg_node_msgs!

closed 2020-07-12 by mikeferguson