ros-drivers/urg_node #43: What could cause an "Abnormal status" error?

venkat-ganesh, 2018-05-24:
Hello,

On Ubuntu 14.04 with ros-indigo, I'm seeing a weird "Abnormal status" error. Here's a screenshot of rqt_robot_monitor -

[image]

Physically, the LIDAR seems to be fine. The LIDAR's LED panel doesn't show any error codes, and I receive data points as expected - 

[image]  [image]

What could cause the URG driver to send out this error? 

Thanks!

anjinkristou, 2018-08-16:
I am interested to know more about this issue. Do you have a pattern to reproduce this issue?

reinzor, 2020-02-07:
We had contact with `Hokuyo` and the reason for this is that this status is only fetched once (when the scanner was not initialized properly by the safety system).

PR #56 (2020-03-14): fix(updateStatus): Update status on diagnostics update
Otherwise the diagnostics information does not really reflect the device
status. Fixes #43
closed 2020-03-14 by tonybaltovski