ros-drivers/urg_node #48: hokuyo UST-10lx LiDAR data not consistent

ArtlyStyles, 2018-12-25:
Hi: I am using hokuyo UST-10LX LiDAR on my robot moving in an indoor environment. I found the data is not consistent. At one moment, the data is like the left part of the image below:

[http://artlystyles.com/tmp/pp.JPG](url)

The robot is the green bar in the image, moving at a slow speed, about 10mph, toward the bottom of the image. Yellow dots are LiDAR points. One side of the wall is pointed by the red array and is at the correct location in this image. The wall is kind of at the back of the robot. The LiDAR has 270 degree view.

At the next moment, however, the data is like the right part of the image above. The wall location is too close to the robot and is totally wrong.

Anyone knows why?

Ainhoamm1, 2022-03-23:
Hi! Did you solve it?

mikeferguson, 2025-05-23:
Reference image is no longer available, closing due to inactivity

closed 2025-05-23 by mikeferguson