ros-drivers/urg_node #55: Could not open network Hokuyo

Miaowaaaa, 2020-01-11:
When I try to run 
`rosrun urg_node urg_node _ip_address:=192.168.0.10`, 
it outputs
`[ERROR] [1578732088.876290125]: Error connecting to Hokuyo: Could not open network Hokuyo:
192.168.0.10:10940
could not open ethernet port.`
I am sure the ip address `192.168.0.10` could be pinged.
And I also tried to modify the local ip address to `192.168.0.XXX`, netmask `255.255.255.0`, it could not work either.
Do some buddies give some help?

Miaowaaaa, 2020-01-11:
I think I have find the troubleshooting tip！I did a totally wrong operation as follows:

- When the Hokuyo laser connects to the computer, I modified the ip address of the network connection to `192.168.0.10`. 

Actually, we should just modify the ip address to `192.168.0.xxx`, except `192.168.0.10`. The reason is that we should keep this network connection in tha same segment. Moreover, the netmask should be set to `255.255.255.0`.

closed 2020-01-11 by Miaowaaaa

muakcal, 2022-12-16:
Would you elaborate on this? I followed the exact same steps you mentioned. However, I am still getting the `[ERROR] [1671229104.984991073]: Error connecting to Hokuyo: Could not open network Hokuyo:
192.168.0.15:10940
could not open ethernet port.`

GetOverMassif, 2023-10-23:
> I think I have find the troubleshooting tip！I did a totally wrong operation as follows:
>
> * When the Hokuyo laser connects to the computer, I modified the ip address of the network connection to [...]

Hello, what do you mean by "modify the ip address", the programme or the hardware? And where to set netmask when using urg_node?

Sitthinan2001, 2024-02-16:
As mentioned, you should change the IP address of the network connection associated with the Hokuyo laser to 192.168.0.xxx (excluding 192.168.0.111) and set the netmask to 255.255.255.0.

Here are the steps you can follow:

Open the window to edit the network connection:

In Ubuntu 18.04, click on the network icon at the top right corner of the taskbar.
Select "Edit Connections" to access network connection settings.
Choose the correct connection:

In the "Network Connections" window, select the connection linked to the Hokuyo laser.
Edit the IP address and netmask:

Change the IP address of the network connection to 192.168.0.xxx (e.g., 192.168.0.12) where xxx is any value different from 111.
Set the netmask to 255.255.255.0.
Save the changes:

Click the "Save" button to save the modifications.
Test the connection:

Open a terminal window and use the ping command to test the connection to the Hokuyo laser:
bash
Copy code
ping 192.168.0.xxx
(where xxx is the IP address you set)
Test using rosrun urg_node:

Once the network connection is correctly configured, try running rosrun urg_node again:
bash
Copy code
rosrun urg_node urg_node _ip_address:=192.168.0.xxx
(where xxx is the IP address you set)