ros-drivers/urg_node #60: Stopped working after updating to newest release

bjsowa, 2020-03-26:
The urg_node seems to not work with Hokuyo URG-04LX on version 0.1.12. This is the output I get (on ROS Kinetic and Melodic):
```
[ INFO] [1585259043.693278963]: Connected to serial device with ID: H1100676
[ INFO] [1585259044.886972703]: Streaming data.
[ WARN] [1585259047.377126484]: Could not grab single echo scan.
[ERROR] [1585259054.246105277]: Error count exceeded limit, reconnecting.
Segmentation fault
```
Not a single message on `/scan` topic is sent.

The 0.1.11 version works fine for me.

mikeferguson, 2020-03-27:
Same here. Caused by https://github.com/ros-drivers/urg_node/pull/56 - if I revert that, things work again.

Interestingly, if I call the update_laser_status service when running 0.1.11, I can reproduce nearly this same failure - I get "could not grab single echo scan" then "error count exceeded limit" - but then the node reconnects and continues publishing data (there is a span of time where the scan topic stops).

tonybaltovski, 2020-03-27:
@reinzor I reverted #56.  I don't have access to any Hokuyo lasers for the time being but will try to investigate it when I can.

amaldo, 2020-03-30:
I just want to confirm that version 0.1.11 was working fine in our lab, but 0.1.12 stops the streaming of laser information about once a second for a short period of time. This is on a Hokuyo UAM-05LP connected over Ethernet.  I just compiled from source and the driver works again fine. 

I guess that calling updateStatus() is not working right, probably interrupting the laser data transmission until it times out.

nickvaras, 2020-03-31:
> I just want to confirm that version 0.1.11 was working fine in our lab, but 0.1.12 stops the streaming of laser information about once a second for a short period of time. This is on a Hokuyo [...]

Same here.

reinzor, 2020-04-01:
Thanks for the update. Strangely enough, this has been tested on our robots with a `Hokuyo UAM-05LP` (and is still running as far as I know) and appeared to work fine. I will come back to this when I've the chance to look into this.

nickvaras, 2020-04-01:
> Thanks for the update and sorry for the inconvenience. Strangely enough, this has been tested on our robots with a `Hokuyo UAM-05LP` (and is still running as far as I know) and appeared to work fine. I will come back to this when I've the chance to look into this.

Communicating over ethernet or usb? On a different note, have you ever seen an Error 51 on those scanners? Hokuyo people suggested it could only be caused by using the scanner on [single message mode (request each scan independently) instead of continuous mode](https://github.com/ros-drivers/urg_c/blob/master/current/src/urg_sensor.c#L907)  , but as far as I can tell, it defaults to continuous mode. It may warrant opening a new issue, except that I'm not even sure if it's a software thing or it's an internal hardware issue with the scanners.

reinzor, 2020-04-02:
> Communicating over ethernet or serial?
ethernet
> have you ever seen an Error 51 
nope

tonybaltovski, 2020-04-08:
Closing this since `0.1.13` has been released to fix the issue.

closed 2020-04-08 by tonybaltovski

sgk-000, 2021-04-24:
Same problem occurred with Hokuyo URG-04LX on version 0.1.11 and 0.1.13 and 0.1.15 on ROS melodic.
This is the output of the error.

`[ INFO] [1619249776.831295975]: Connected to serial device with ID: receive error.
[ INFO] [1619249776.940173001]: Streaming data.
[ WARN] [1619249778.593896236]: Could not grab single echo scan.
[ERROR] [1619249784.985762941]: Error count exceeded limit, reconnecting.
[ INFO] [1619249787.372052135]: Connected to serial device with ID: receive error.
[ INFO] [1619249787.480759819]: Streaming data.
[ WARN] [1619249789.136214096]: Could not grab single echo scan.
[ERROR] [1619249795.527071800]: Error count exceeded limit, reconnecting.
[ INFO] [1619249797.913377470]: Connected to serial device with ID: receive error.
[ INFO] [1619249798.026012049]: Streaming data.
[ WARN] [1619249799.677856377]: Could not grab single echo scan.
[ERROR] [1619249806.068363528]: Error count exceeded limit, reconnecting.
[ INFO] [1619249808.454476978]: Connected to serial device with ID: receive error.
[ INFO] [1619249808.561685403]: Streaming data.
[ WARN] [1619249810.217528696]: Could not grab single echo scan.
[ERROR] [1619249816.609601278]: Error count exceeded limit, reconnecting.`

sgk-000, 2021-04-24:
Also same problem occurred with URG-04LX on foxy-devel.
I can not receive any msg in /scan topic