ros-drivers/urg_node #66: How do call "UrgNode::run()" in latest code on ros2-devel branch

n-hamaike, 2020-05-11:
I'm trying to test "LaserScan Visualization" following this link(https://index.ros.org/p/urg_node/#dashing), but it doesn't work.
The result of debugging, "UrgNode::run()" is not called.(I set break point by GDB but not stopped, and not called from anywhere in source code)

With refering to commit log, I guess it seems disappeared when "urg_node" modified to component node.
(https://github.com/ros-drivers/urg_node/pull/50/commits/ad15f2354307df77c4098c71f26ce361b9b10572)

I added "UrgNode::run()" caller, then it works.

In latest code on ros2-devel branch, How do call "UrgNode::run()"?
"UrgNode::run()" caller is not necessary?

I'd appreciate it if someone could answer my questions.

mikeferguson, 2020-06-07:
So, here's the hack I added, seems to work (not sure if it breaks other compositions though): https://github.com/mikeferguson/urg_node/commit/f308eee4f6a567a01077f205db9d130124f564e0

n-hamaike, 2020-06-10:
@mikeferguson
thanks for your comments and fixes!