ros-drivers/urg_node #68: No executable found with `ros2 run urg_node urg_node` command on ROS 2 dashing

ShotaAk, 2020-06-05:
# Describe the bug

I ran a command `ros2 run urg_node urg_node` on ROS2 dashing environment.
That results the message `No executable found`.

# Environments
- OS: Ubuntu 18.04 
- Software Version: latest of ros2-devel branch (cb0efdc61c3bb4fde59ce3d1ec8b6bf1cc6da5dc) 
- LiDAR: URG-04LX-UG01

# To Reproduce

1. Clone this repo locally and checkout to ros2-devel branch
2. Run `colcon build --symlink-install` then run `source install/setup.zsh`
3. Run `ros2 run urg_node urg_node`
4. See error `No executable found`

# Expected behavior

The `urg_node` will be running then publishing `/scan` topics.

# Additional context

The correct executable node name is `urg_node_driver` that CMakeLists.txt shows.

https://github.com/ros-drivers/urg_node/blob/cb0efdc61c3bb4fde59ce3d1ec8b6bf1cc6da5dc/CMakeLists.txt#L54-L56

So, I ran a command `ros2 urg_node urg_node_driver` but no /scan topic published.
This problem same as https://github.com/ros-drivers/urg_node/issues/66.

ShotaAk, 2020-06-05:
I tried to make an executable node that executes UrgNode.run().
The node publish `/scan` topic successfully.

**Source code of the executable node (urg_node_executable.cpp)**
```cpp
#include <memory>
#include "rclcpp/rclcpp.hpp"

#include "urg_node/urg_node.hpp"

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::executors::SingleThreadedExecutor exec;
  rclcpp::NodeOptions options;
  auto urg_node = std::make_shared<urg_node::UrgNode>(options);

  urg_node->run();

  exec.add_node(urg_node);
  exec.spin();
  rclcpp::shutdown();
}
```

**New lines of the CMakeLists.txt**
```cmake
add_executable(urg_node_executable src/urg_node_executable.cpp)
target_link_libraries(urg_node_executable urg_node)
ament_target_dependencies(urg_node_executable
  rclcpp)

install(TARGETS urg_node_executable
  DESTINATION lib/${PROJECT_NAME}
)
```

( Actually, I wanted to change names `urg_node` to `urg_node_component`, `urg_node_executable` to `urg_node`.)

To reproduce:
1. Run ` ros2 run urg_node urg_node_executable  __params:=$HOME/ros2_ws/src/urg_node/launch/urg_node_serial.yaml`
2. Run `ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 "world" "laser"`
3. Run `rviz2`

Result:
[image]

mikeferguson, 2020-06-09:
So, it looks dashing has version 1.0 released into it - the node has been renamed to "urg_node_driver". I've opened a PR to fix the documentation: https://github.com/ros-drivers/urg_node/pull/70

It should be noted that the node still doesn't run because of #66 - see my patch there (I haven't opened a PR yet for that one, cause I'm not sure it's the "right" fix).

mikeferguson, 2020-06-09:
See also #71 - which should fix the run() issue in #66

mikeferguson, 2020-06-11:
1.0.1 released to foxy: https://github.com/ros/rosdistro/pull/25425

ShotaAk, 2020-06-12:
@mikeferguson thank you for fixing and updating the package!