ros-drivers/urg_node #76: Noetic: segfault after connection success with package from APT

Mindstan, 2020-09-05:
I have installed `urg_node` from APT (`ros-neotic-urg-node`), but when I run `rosrun urg_node urg_node _serial_port:=/dev/ttyACM1` it crashes after the connection to the device:
```
[/urg_node: INFO] [1599296638.701332302] : Connected to serial device with ID: H1102605
[/urg_node:DEBUG] [1599296639.596760024] : Unable to update reconfigure limits. Not Ready.
[1]    30627 segmentation fault (core dumped)  rosrun urg_node urg_node _serial_port:=/dev/ttyACM1
```

However, if I clone this repository and build myself the package it works smoothly :
```
[/urg_node: INFO] [1599295313.022726465] : Connected to serial device with ID: H1102605
[/urg_node: INFO] [1599295314.117096183] : Streaming data.
```

- OS : Ubuntu 20.04
- Device : URG-04XL-UG01
- urg_node version: 0.1.14
- urg_c version: 1.0.405 (from APT ros-neotic-urg-c)
- GCC version: 9.3.0-10ubuntu2

GDB info:
```
Starting program: /opt/ros/noetic/lib/urg_node/urg_node 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4981700 (LWP 32503)]
[New Thread 0x7fffeffff700 (LWP 32504)]
[New Thread 0x7fffef7fe700 (LWP 32505)]
[New Thread 0x7fffeeffd700 (LWP 32506)]
[/urg_node: INFO] [1599297220.564435432] : Connected to serial device with ID: H1102605
[New Thread 0x7fffee7fc700 (LWP 32509)]
[New Thread 0x7fffedffb700 (LWP 32510)]

Thread 7 "urg_node" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffedffb700 (LWP 32510)]
0x00007ffff7c566bd in std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) backtrace
#0  0x00007ffff7c566bd in std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff79522b5 in urg_node::URGCWrapper::start (this=0x7fffdc0010a8) at ./src/urg_c_wrapper.cpp:870
#2  0x0000000000000000 in ?? ()
```

bostoncleek, 2020-09-24:
I am running this node with the same environment setup as @Mindstan. I am attempting to communicate with the Hokuyo via Ethernet and observe the same error. 

```gdb
Starting program: /opt/ros/noetic/lib/urg_node/urg_node scan:=front/scan __name:=urg_front __log:=/home/administrator/.ros/log/6d7d83e4-fe99-11ea-9aea-abb0e9e33a90/ridgeback-urg_front-2.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4996700 (LWP 39609)]
[New Thread 0x7fffeffff700 (LWP 39610)]
[New Thread 0x7fffef7fe700 (LWP 39611)]
[New Thread 0x7fffeeffd700 (LWP 39612)]
[ INFO] [1600974552.380452838]: Connected to network device with intensity and ID: H1880745
[New Thread 0x7fffee7fc700 (LWP 39613)]
[New Thread 0x7fffedffb700 (LWP 39614)]

Thread 7 "urg_node" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffedffb700 (LWP 39614)]
0x00007ffff7c6a6bd in std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) backtrace
#0  0x00007ffff7c6a6bd in std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff79662b5 in ?? () from /opt/ros/noetic/lib/liburg_c_wrapper.so
#2  0x0000000000000000 in ?? ()
```

bostoncleek, 2020-09-24:
I found that building from source only works if the build type is Debug or RelWithDebInfo. At compile time I get two warnings. 

```gcc
/home/administrator/sawback_ws/src/urg_node/src/urg_c_wrapper.cpp: In member function ‘bool urg_node::URGCWrapper::setSkip(int)’:
/home/administrator/sawback_ws/src/urg_node/src/urg_c_wrapper.cpp:872:1: warning: no return statement in function returning non-void [-Wreturn-type]
  872 | }
      | ^
/home/administrator/sawback_ws/src/urg_node/src/urg_c_wrapper.cpp: In member function ‘std::string urg_node::URGCWrapper::sendCommand(std::string)’:
/home/administrator/sawback_ws/src/urg_node/src/urg_c_wrapper.cpp:593:8: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
  593 |   write(sock, cmd.c_str(), cmd.size());
```

Looks like ```cpp bool URGCWrapper::setSkip(int skip) ``` on line 868 of urg_c_wrapper.cpp as no return. 

I am not sure if this solution is correct but I believe the parameter skip to be either 0 or 1 so I return skip. This currently prevents the node from crashing. I would like to hear back from a maintainer on this. As more people switch to noetic this is likely to become a more noticed issue.

bostoncleek, 2020-09-24:
I found that the node still crashed when I build from source if the build type is release. If the build type is debug there is no issue. However please see the PR.

mikeferguson, 2025-05-22:
Noetic is EOL tomorrow