ros-drivers/urg_node #79: ros2 urg_node parameter file

edfang1, 2020-11-07:
installed sudo apt install ros-foxy-urg-node.
ros2 run urg_node getID works fine and returns ID.
ros2 run urg_node urg_node_driver __params:=/opt/ros/foxy/share/urg_node/launch/urg_node_ethernet.yaml

returns can't find serial port 115200 /dev/cu.usbmodemxxxxx

I've set the ip address in the yaml file and it just doesent seem to be reading the paramtere file anywere.  Any idea on why its not readin the parameter file and using the IP address of the hokuyo (since thats the model I have) ?

Tried to git clone and build, but colcon build comes out with a bunch of catkin (ros1) type of errors.  

Thanks for any assistance.  Apologies if this is an idiot request.  I've read everything I could find and search upon.  Thanks

mikeferguson, 2020-11-07:
> ros2 run urg_node urg_node_driver __params:=/opt/ros/foxy/share/urg_node/launch/urg_node_ethernet.yaml

You'll have to post the yaml file if there is going to be any hope of debugging this

> Tried to git clone and build, but colcon build comes out with a bunch of catkin (ros1) type of errors.

Have you checked out the right branch? This repo is used for both ROS1 and ROS2.

edfang1, 2020-11-07:
Updated the original file.  the orignal is run with urg_node being installed as a binary (apt install).  I realized that I was pulling from the wrong branch last night as it dawned on me .. . lol.  I'll try building it from source, but was wondering why the binary one wasnt able to take the param file and seemingly was ignoring it.

mikeferguson, 2020-11-07:
Ok - since we generate the node using rclcpp_components_register_node - can you try changing the node name to "urg_node_driver" in your yaml file? I think that ends up being the default node name (but I don't seem to have any example of loading from a yaml -- I've been putting the params directly into a launch file)

edfang1, 2020-11-08:
tried changing the yaml file to urg_node_driver and its the same.  Like its not even picking up the param file on the call.

I will eventually be doing it from a launch file, but right now just testing the new hoyuko with rviz - so that would be the first step.  Not sure if its my reference or whether its just not picking up the param declaration on the command line.

Kakcalu13, 2021-04-05:
> tried changing the yaml file to urg_node_driver and its the same. Like its not even picking up the param file on the call.
>
> I will eventually be doing it from a launch file, but right now just [...]

Did you manage to get it fixed? If so, what was the solution?

Kaetera, 2021-04-23:
I encountered the same error with ROS2 Foxy on Ubuntu.
The run instructions provided in the Readme of the foxy/devel branch seems out-of-date, the correct way to pass the YAML parameters file and was:

`ros2 run urg_node urg_node_driver --ros-args --params-file src/test/urg_params.yaml`

Else the YAML file is silently ignored and not IP was set.
This is the way according to ROS2 doc: https://github.com/ros-drivers/urg_node/issues/79?_pjax=%23js-repo-pjax-container

PR #112 (2024-01-31): README syntax updates
Updates to ROS 2 syntax and minor rewording. I also moved RViz to the visualization section.
Fixes #79
commit 369ee4d8 (2024-01-31): README syntax updates (#112)

Updates to ROS 2 syntax and minor rewording. I also moved RViz to the
visualization section.
Fixes #79
closed 2024-01-31 by mikeferguson