ros-drivers/urg_node #81: uam-05lp stops publishing data

femtogram, 2020-12-08:
I've been running into an issue with urg_node stops publishing data without spitting out any errors when running with uam-05lp.

Would a bandwidth issue cause something like this? I have 3 urg-20lx and 2 uam-05lp on the machine. the uam-05lp are the only ones this seems to happen with.

In order for it to start back up again, i need to kill and restart the uam-05lp rosnodes.

Any thoughts as to what might be going on? Thanks!

comwise, 2021-08-02:
you can see this [https://github.com/comwise/urg_node](url)

comwise, 2022-04-26:
> Hi comwise, the link you provided is broken. I am interested by this issue as I am also using uam scanner. kotaongit did you manage to find the problem?

yes, you can direct to this [https://github.com/comwise/urg_node](url)