ros-drivers/urg_node #82: How to disable interpolation for the sensor data (ros2-devel branch)

JaehyunShim, 2020-12-21:
As the title says, how can I disable interpolation for the sensor data? Also, if possible, can you add a parameter for setting the on/off state of interpolation for those users who need raw sensor data?

Jaehyun

mikeferguson, 2020-12-21:
I'm not aware that this driver does any interpolation - can you post some screenshots or data that shows what interpolation you're seeing?

JaehyunShim, 2020-12-22:
@mikeferguson 

In the picture below, you can see these noise data (red circles) presumably created by interpolation around the obstacles (blue circles). 
[image]

You can see these dots connecting the obstacles (blue circles) more clearly in the next picture. 
[image]

JaehyunShim, 2020-12-22:
FYI, UAM-05LP-T301 is what I am testing with.

mikeferguson, 2020-12-22:
These are not "interpolation" - this is what is known (at least in ROS-land) as a "shadow" - and it's an artifact of the laser operation - not something the drivers do.

The ROS1 package "laser_filters" includes a shadow filter to remove these points (as was popularized on the PR2). The package is semi-ported to ROS2, I had previously tried to run it (about 3 months ago) but ran into issues at the time.

mikeferguson, 2020-12-29:
Closing due to lack of response - please reopen if you find additional information that shows these are not sensor artifacts.

closed 2020-12-29 by mikeferguson